arduino/mkrwan1300-fw #3: Library level control for ADR

dan0h, 2018-05-03:
Hi :) Just wondering if there is anything documented on how you can activate/deactivate ADR in the firmware? I'm trying to use the MKR1300 as a TTN Mapping node, but it appears as though ADR is activated by default, and as such its scaling its own spread factor, and I would like to be able to lock the MKR1300 at SF7 for mapping purposes, has this been implemented yet?

dan0h, 2018-05-03:
Awesome, many thanks, exactly what I need!

dan0h, 2018-05-03:
Works perfectly, ADR now appears to be off. Though I still can't find a way to manually set the spread factor down to SF7, have looked around in the file linked above (at.c) but can't find anything related to SF, Spread, or Factor. ADR definitely turned off though, just now its locked at SF12!

dan0h, 2018-05-03:
That works perfect for me too, thank you for your time :) The only oddball part of the behavior left for me is that I cannot set the SF with ADR set to False, every uplink seems to generate a downlink that I'm not requesting - is this TTN/LoRa negotiating ADR? Thanks for your time, really appreciated, this device is a lot more involved than a LoPy!

facchinm, 2018-05-03:
@aschr @xoseperez thanks a lot for your effort! If you could open a couple of PRs on MKRWAN library repo I'll be super glad to merge them!

aschr, 2018-05-04:
@facchinm i just made (my first :-) ) PR on the MKRWAN lirary.

facchinm, 2018-05-04:
And I just merged it :smile: Thanks  for contributing! The updated library should be available via Library manager in the next few hours

dan0h, 2018-05-04:
Great work all around, thanks to all of you :)

facchinm, 2018-05-08:
Closing by publishing MKRWAN library v 1.0.6

closed 2018-05-08 by facchinm