arduino/mkrwan1300-fw #5: Request:  Command to set sub bands for US915 region

sslupsky, 2018-09-03:
I cannot find a way to set the sub bands.  That is, to enable and disable the various channels in the US915 region.  This is required for the US915 region where any particular gateway likely only has 8 of the 64 regional channels enabled.  In some cases, people are operating single channel gateways.

Also, I see two definitions for the US915 region.  "US915" and "US915_HYBRID"  However, there is no documentation that I could find that explained what the difference is.

sslupsky, 2018-09-16:
@facchinm The US915_HYBRID region seems to work more reliably.  However, I cannot find any information on what channels this sub band supports.  Since it seems to work reliably I think its channels 0 to 7 and channel 64.  But I can't be sure.  Do you know?

facchinm, 2018-09-17:
Hi @sslupsky , the situation is not very clear with LoRa stack; the latest branch from ST (ported here https://github.com/facchinm/mkrwan1300-fw/tree/master-1.2.0) includes the US915 patch but doesn't seem to be able to connect to an actual network; I'll keep you posted if there are news.