arduino/mkrwan1300-fw #6: Cannot set DataRate when ADR is not set

AmedeeBulle, 2018-09-04:
It seems that the DataRate gets overwritten and forced to DR0 (SF12BW125) when ADR is not set.

I have a small sketch which sends console input to the modem to illustrate the problem:
```
Module version is: ARD-078 1.1.6
Successfully joined TTN
Enter command mode...

AT+ADR?
+OK=0
AT+DR?
0
AT+DR=5
+OK
AT+DR?
5
AT+SEND=1:*
+OK
AT+DR?
0
```
One can see that after the `send` the DataRate is reset to 0, and I can confirm from the TTN console where the packet is sent to that packet was sent with DR0 (SF12BW125).

When ADR is enabled, DataRate selection is working well -- e.g.:
```
AT+ADR=1
+OK
AT+ADR?
+OK=1
AT+DR=5
+OK
AT+DR?
5
AT+SEND=1:*
+OK
AT+DR?
5
```
DataRate stays at 5 and  the TTN console reports DR5 (SF7BM125)

sslupsky, 2018-09-14:
Hi @AmedeeBulle Would you be willing to share your sketch for command mode?  That looks very useful.

AmedeeBulle, 2018-09-15:
It is very simple, but if it can help I just uploaded it: [AmedeeBulle/TtnModemTest](https://github.com/AmedeeBulle/TtnModemTest)

sslupsky, 2018-09-16:
@AmedeeBulle Indeed you are correct, quite simple!!!  Thanks for posting!  This will help me debug the MKR WAN a bit better.  Have you had any success with low power consumption?  I am unable to get the MKR WAN 1300 below 1.16mA during sleep and would appreciate to know if you have been able to reduce the power further than that?

flhofer, 2021-06-21:
@AmedeeBulle fixed with f849548ab7f7f77bdf6179708103a414aaf2efb7 (https://github.com/arduino/mkrwan1300-fw/pull/35)

The data rate used in non-ADR was always overwritten with an Init value at firmware startup. The changes patch this by updating the init parameter and set also the channel default DR. The latter must happen, as internally the send function checks for max length against the default if ADR is turned off.
The next release (1.3.x) should not have any of these issues. 😃

jpmeijers, 2021-08-11:
@flhofer should this fix already be available? I'm testing the MKR WAN 1300 for The Things Summer Academy, and this bug is causing issues for my workshop.

rqg0717, 2022-01-28:
any update on this please? facing the same issue... thank you

flhofer, 2022-01-31:
Hi, @rqg0717 

It is fixed in not yet released [v1.2.4 of my pull request](https://github.com/arduino-libraries/MKRWAN/pull/93). It includes the necessary modem firmware update in the MKRWAN standalone sketch. (click on the link to the repo next to the title)