arduino/mkrwan1300-fw #14: MKR LoRa wan 1300 as a gateway

pmerlin13, 2019-03-04:
Hi,

I'm new using LoRa.

I got one LoRa mkrwan 1300 and i'm wondering if i can use it as a gateway just to read data from end devices.

Best regards,
Merlin Pineda

facchinm, 2019-03-06:
Hi @pmerlin13 ,
yes you can (but there's no code to do it) :smile: 
Jokes apart, the setup you wont to achieve is usually called "single channel gateway" which works just for very small setups (all boards must be transmitting on the same channel with different timings).

The firmware we ship doesn't provide this functionality, but combining the useful parts of https://github.com/things4u/ESP-1ch-Gateway-v5.0 and https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/pull/57 will do it (it would be a very nice project).

pmerlin13, 2019-03-06:
Hi,

Is there any way to send AT command to the MKR wan 1300?

pmerlin13, 2019-03-12:
@facchinm have use use AT commands to set the Device EUI and AppKey of the Mkrwan 1300.
I have try multiple ways and got nothing. 
Example : AT+APPKEY

pedros89, 2022-06-11:
Hello,
I have a a few [analog sensors from Adeunis](https://www.adeunis.com/en/produit/analog-wired-sensor-interface/)
That are already connected to a gateway via ABP connection. I have all their addresses, NWS KEY and LORA APP S KEY which vary for each sensor.
Basically I would need to see what they are sending and use MKR Lora 1300 to activate some relays.
I can connect to their very same network but I cannot read what they are sending. Did you find a way to do it?
Thank you
Pietro