arduino/mkrwan1300-fw #15: Problem with the JOIN procedure

mjunior-fitec, 2019-05-07:
@facchinm I've been experiencing some troubles with the JOIN procedure using the MKRWAN.h library. To confirm I used a sketch that enables me to send AT commands directly to the modem.
On those tests I realized that the modem doesn't answer the AT+JOIN command. It only returns something when it receives the JoinAccept message from the Network Server. The MKRWAN.h implements a 1 minute timeout, waiting for the "EVENT=1,1" message. When everything runs fine it works, but it's quite common that the JoinAccept message takes 2 or 3 minutes to arrive. Also, when something is wrong (no gateway reachable, wrong keys...) it looks like the modem keeps sending JoinRequest indefinitely! On my tests I saw it runnig for about 2 hours. When it occurs, the device may went in a loop, trying to establish a _join_ .
It might be a big problem, since the JoinAccept message can be already sent, the radio received it correctly, the keys were set and the "EVENT=1,1" message sent but never received by the Arduino processor. In this situation, when the new "AT+JOIN" message is sent, the modem doesn't answer and, **the worst**, the radio sent nothing! I believe that the modem must answer all the commands, everytime an "AT+JOIN" message is received by the modem it should answer, for example with a "+OK", and when the join was already done and the Join State is JOINED it should answer informing this situation (Already joined?).
Checking the Murata documentation, they say that on their modem implementation this is the behaviour. Also, there are many other usefull AT commands like the Multi Cast configuration, wich would be very usefull on the MKRWAN1300.
Does the current ST implementation have some improvement related to it?
Are you planning to correct this on the modem and in the MKRWAN.h lib?

Thanks.

facchinm, 2019-05-08:
Hi @mjunior-fitec ,
while we tried to make the ST stack as similar as possible to the murata one, it was probably not a great idea :slightly_smiling_face: 
Currently we are working on a fw and a library that will supersede the actual ones, solving most of the reliability problems and giving access to the most recent LoRa specifications.

mjunior-fitec, 2019-05-10:
@facchinm can you tell me what is the expectation to have a Multicast implementation on MKRWAN1300, just like `AT+MCAST` that exists on the Murata firmware?

terryzar, 2021-02-05:
Hello Facchinm, I am having same issue where I need to ad MCAST or CHMASK to tthe code.  Do you have the Murata AT command documentation for these commands?  Much appreciated!

facchinm, 2021-03-29:
:wave: 
We added support for channle masks in the lates firmware (1.2.3) and in this PR https://github.com/arduino-libraries/MKRWAN/pull/90 . Would you mind testing it and reporting if it works? Thanks!

terryzar, 2021-03-30:
> <http://www.linkedin.com/company/unimar?trk=hb_tab_compy_id_884686>[image:
> VOSB Logo]
>
> 3195 Vickery Rd. | North Syracuse| New York| 13212
>
> (315) 699-4400 | ex. 309 | www.unimar.com
>
> This email and any attachments (“Communication”) is strictly confidential
> and intended solely for the addressee.  It may contain information which is
> confidential and/or covered by legal, professional or other privilege.  If
> you have received this Communication in error, you must not disclose, copy,
> distribute or take any other actions in relation to it.  Instead, please
> inform us as soon as possible on (315) 699-4400.
>
> *From:* Martino Facchin ***@***.***
> *Sent:* Monday, March 29, 2021 10:09
> *To:* arduino/mkrwan1300-fw
> *Cc:* terryzar; Comment
> *Subject:* Re: [arduino/mkrwan1300-fw] Problem with the JOIN procedure (#15)
>
> 👋

Hello:

I will definitely test it Facchin!   I will likely not be able to do so for
another 2 weeks, but will do so!

Thank you!

Best Regards,

Terry Zarnowski

*Terry Zarnowski*

this PR arduino-libraries/MKRWAN#90

flhofer, 2021-04-22:
> 👋
> We added support for channle masks in the lates firmware (1.2.3) and in this PR [arduino-libraries/MKRWAN#90](https://github.com/arduino-libraries/MKRWAN/pull/90) . Would you mind testing it and reporting if it works? Thanks!

Hi! I will have the opportunity to fully test the channel mask in the next few days. My gateway just gave up and I hope to get a new one by tomorrow afternoon. If I remember right, when writing the code I already adjusted something in the arduino library. Will post it here as soon as I get some results.

flhofer, 2021-05-02:
@facchinm @terryzar 
HI, the channel mask works with my pull request version https://github.com/arduino-libraries/MKRWAN/pull/93
The thing I noticed is:
If you use ABP, the modem does not know anything but the default frequencies. You might need to compile with `USE_SEMTECH_DEFAULT_CHANNEL_LINEUP   1` to use all 8 standard channels. 
Using OTAA it learns from the Network server, as expected. BUT You can not disable the default channels (1-3)

@mjunior-fitec 
Also, with my latest library and firmware version, the modem responds with `+ERR_BUSY` if the join goes in timeout and you try to send. So, if it takes longer you can just poll the join status and then continue. 
Also, the firmware is implemented as described. If the modem has joined, it should actually return immediately after a join. In my test setup I join the network multiple times in a few minutes over OTAA and the modem typically answers within a 3-5 second span. Try the version in my pull request. If it still doesn't work, it could be that you have problems with your network server (had something similar)

Best,
Florian

terryzar, 2021-05-03:
Hi Florian:

Thank you for doing this work, much appreciated!

Best Regards,

Terry Zarnowski

*Terry Zarnowski*

*From:* Florian Hofer ***@***.***
*Sent:* Sunday, May 02, 2021 15:38
*Cc:* terryzar; Mention