arduino/mkrwan1300-fw #20: AT response codes do not conform to V.250 standard AT command format

sslupsky, 2020-03-29:
The AT response codes all begin with "+OK".  This is not correct according to the V.250 standard for AT commands.  Section 5.7.2 reads:

> Extended syntax result codes shall be prefixed by the "+" character to avoid duplication of basic format result codes specified in TIA-602 and by manufacturers. Following the "+" character, the name of the result code appears; result code names shall follow the same rules as command names (see 5.4.1). It is strongly advised that the reservation of the first character of command names noted in Appendix I also be observed with regard to the assignment of names of extended syntax result codes.
> Extended syntax result codes may include the reporting of values. The definition of the result code shall specify whether or not values are appended to the result code, and, if so, how many, their types, and their assumed default values if omitted. When no values are to be reported, the result code appears in the simplest form:
> +\<name\>
> If a single value is to be reported, the form of the result code shall be:
> +\<name\>: \<value\>
> Note that a single space character (ASCII 20h) separates the colon character (ASCII 3Ah) from the \<value\>; no space appears between the result code name and the colon. If multiple values are to be reported with the result code, the form is:
> +\<name\>: \<compound_value\>
> where \<compound_value\> follows the rules specified in 5.4.2.3.

flhofer, 2021-05-02:
Hi @sslupsky 
https://github.com/arduino/mkrwan1300-fw/pull/35 should do the trick :) I tried to adapt it a bit to +<name>=<value>, which's closer and less ambiguous. You, however, suggest using ": " instead, i.e., space and colon? When creating this AT slave, I figured that Semtech just did a thing on their own, a lookalike. Also, if you read the standard, there would be a lot more to change, e.g., mandatory commands such as `+ATZ` are missing. While having unambiguous return strings makes sense, I don't think that a strict standard implementation would be that useful as most people will use the library methods.

What do you think?
Best,
Florian

sslupsky, 2021-05-06:
Hi @flhofer .  Thank you for looking into this issue.  I believe the reason the standard used a semicolon instead of the equals sign is because the response is acknowledging the assigment versus using the equals sign to make the assignment.  It is a subtle difference but I think it is appropriate to to so.